sipeed/bl602-hal #10: Invalid drop of ClkCfg reference

mkroman, 2021-01-06:
Clippy has [found a bug](https://github.com/sipeed/bl602-hal/actions/runs/465334147) in `clock::Strict::freeze`:

```
error: calls to `std::mem::drop` with a reference instead of an owned value. Dropping a reference does nothing.
   --> src/clock.rs:149:9
    |
149 |         drop(clk_cfg); // logically use its ownership
    |         ^^^^^^^^^^^^^
    |
    = note: `#[deny(clippy::drop_ref)]` on by default
note: argument has type `&mut gpio::ClkCfg`
   --> src/clock.rs:149:14
    |
149 |         drop(clk_cfg); // logically use its ownership
    |              ^^^^^^^
    = help: for further information visit https://rust-lang.github.io/rust-clippy/master/index.html#drop_ref
```

I don't quite understand the point of `ClkCfg` at all - why does it just have an inner *unit type*, and why is it immediately being dropped after being passed as an argument?

luojia65, 2021-01-06:
My original design of this configurator is that it make uses of Rust's ownership system. The ClkCfg struct represents all the registers we should use in freeze procedure. In other HAL crates, HAL library uses this ownership representation to modify the registers logically belong to it. This strucutr shouldn't be dropped in further designs, it should have functions which returns PAC register structures to modify clock vales.
If there's no such field, all the contexts can freeze clock configurators which may result in data races (if I am correct).

mkroman, 2021-01-09:
I see. So the inner unit type in ClkCfg (`()`) is just a placeholder for now?

With regards to the `drop` call - is there any reason for this to be there? We don't take ownership of the type, and we don't borrow anything from it, so the mutable borrow only lives for as long as the call to `freeze` anyway.

luojia65, 2021-01-09:
At first I thought the clock configuration is done by modifying one set of register like stm32, but this chip requires multiple registers from different peripherals. Maybe we can split out more register ownership from each of peripherals, or if possible just remove this ClkCfg struct totally.

mkroman, 2021-01-11:
I'm going to remove the manual drop from `clock::Strict::freeze` in #11 since it's literally a no-op, but I'll of course keep the mutable borrow of the struct for forward compatibility.

mkroman, 2021-01-14:
Fixed in #11

closed 2021-01-14 by mkroman